onitake/gslx680-acpi #4: Chuwi Hi8 support

djvdorp, 2016-01-26:
(RE: https://github.com/Manouchehri/vi8/issues/1)

My codebase: https://github.com/djvdorp/hi8

My master tree contains the windows10 drivers (which I obtained from windows10 myself) and linux drivers (which I downloaded via techtablet). 

The windows10 driver is a SileadTouch.{cat,sys} but no .fw file is incuded with the driver dump I made (but I might be able to dig this up manually).
However, @barni2000 wrote that it is possible to extract a firmware.fw from SileadTouch.sys, but since I have no clue on how to find the firmware begin and end I have not tried that yet.

The linux driver is a SileadTouch.{sys,fw} which I unscrambled via:
```./unscramble SileadTouch.fw TS_CFG.h```
That unscrambled version can also be found in my master tree.
However, after that I kinda get stuck. I don't have an idea which parameters to pass to ```fwtool``` even though I have read the [README](https://github.com/onitake/gslx680-acpi/blob/master/README.md)

I am just preparing these touch firmwares for when I am actually going to install Debian/Ubuntu on the device (or run it via USB OTG). I have not (tried to) run linux on it yet. I am unsure about what driver I am going to use at this point (eg. the gslx680-acpi or rastersoft you mentioned).

Thanks for your time and patience in advance!

onitake, 2016-01-26:
SileadTouch.sys does contain a firmware image (usually), but there is no straightforward way to extract it. You need to open the file in a hex editor and search for the sequence F0 00 00 00. The firmware has a peculiar format that makes it relatively easy to recognize. It's a series of 32-bit address and 32-bit data word pairs, but the address is actually only 8-bit, so there's lots of 00 00 00s there. The F0 00 00 00 is a "select page" command, which is followed by the page number and (usually) 128 address-data pairs. You need to find out where the firmware ends by yourself. If you can't figure it out, I'll extract the fw for you.

If you have anything better, for example a TS_CFG.h or a SileadTouch.fw, use that instead.

The rastersoft driver runs in user space and requires you to figure out which GPIOs yourself. If you use gslx680-acpi, it will obtain that information from ACPI automatically. It's not guaranteed to work, but I've had some success so far.

So, as for fwtool. You don't need to use it if you go for the rastersoft driver, as it uses the legacy format that is contained in SileadTouch.sys. gslx680-acpi uses a different firmware format, so you do need to use it for that. If you have a SileadTouch.fw, you need to either unscramble it using the unscramble tool (for rastersoft), or convert it into a firmware for gslx680-acpi directly using fwtool. No need to unscramble first, in that case, just pass the right parameter (-3).

So, to sum it up:
1. Obtain legacy firmware from SileadTouch.sys using a hex editor, save to gslxxxx.fw (or whatever you like)
    F0 00 00 00 xx 00 00 00
    ...
    7C 00 00 00 yy zz ww vv
2. Convert into silead_ts.fw with fwtool, specify the correct parameters for your touchscreen and controller (these are totally device specific, you may need to experiment or obtain them somewhere else)
    ./fwtool -c gslxxxx.fw -m 1680 -w 940 -h 750 -t 10 -f track silead_ts.fw
3. Copy silead_ts.fw to /lib/firmware
    cp silead_ts.fw /lib/firmware

onitake, 2016-01-26:
The SileadTouch.fw from your repo has the fw image at range 0x9430-0x12edf, I think.

barni2000, 2016-01-26:
I get some error:
[ 2366.225028] gslx680 i2c-MSSL0001:00: gsl_ts_probe: got a device named MSSL0001:00 at address 0x40, IRQ 205, flags 0x0
[ 2366.225227] gslx680 i2c-MSSL0001:00: gsl_ts_probe: error obtaining power pin GPIO resource
[ 2366.233039] gslx680: probe of i2c-MSSL0001:00 failed with error -16

barni2000, 2016-01-26:
I have used to list acpi ids ````ls /sys/bus/acpi/devices/```` there are MSSL0001 and MSSL1680.

onitake, 2016-01-26:
@barni2000 is this related to @djvdorp 's firmware issue?

If not, please post open a new issue. Don't post here.

barni2000, 2016-01-26:
Yes i tried on my friend's chuwi hi8.

onitake, 2016-01-26:
If there's a MSSL1680 as well, I don't think the driver should match the MSSL0001, unless you have two separate touchscreens. Can you send me the DSDT, or decompile it using [these instructions](https://github.com/Manouchehri/vi8/issues/1#issuecomment-129013952) and post the relevant sections?

barni2000, 2016-01-26:
I couldn't send because I gave it back to my friend's.

djvdorp, 2016-01-27:
@onitake and @barni2000 , thanks for all your input and replies. In my first attempt to get Debian booting on this thing, I followed this:
https://www.debian.org/CD/faq/#write-usb

However, when connecting through my ```Acasis Micro USB OTG charging Hub H027``` together with my keyboard (needed to press ESC so I can choose the USB drive to boot from), it doesn't want to boot from the USB as it doesn't seem to recognize it or it's contents. Can you guys link/explain me what the easiest way is to get a Linux Live USB running on there (or MicroSD card) so I can gather you the information you requested, as root?

onitake, 2016-01-27:
I don't know if you can boot from µSD at all. Some cheap tablets have crabby UEFI BIOSes that don't support the SD card interface. USB boot through a hub should work, though.

The CD image you'd like to use will not work though, because it only includes a 64bit EFI boot loader. Baytrail tablets usually contain a 32bit UEFI BIOS, so you need a 32bit version. On the other hand, Debian i686 does not contain any EFI boot loader at all. 

What you need is the Debian multiarch installer, which contains any boot loader you could possibly need, legacy BIOS, El Torito, 32bit (U)EFI, 64bit (U)EFI. It will even boot on older Intel Macs successfully. Unfortunately, there is no multiarch Live CD variant, so will need to install Debian first.

Use this: http://cdimage.debian.org/pub/debian-cd/current/multi-arch/iso-cd/
Or this: http://cdimage.debian.org/pub/debian-cd/current/multi-arch/iso-dvd/

djvdorp, 2016-01-28:
Thanks for your information (again) @onitake. I managed to get Ubuntu 15.10 running from a Live USB with the excellent instructions from @Manouchehri here:
https://github.com/Manouchehri/vi8/blob/master/Ubuntu_instructions.md

I just ran a few quick commands on it, the most interesting one for now was ```ls /sys/bus/acpi/devices/``` which only gave me ```MSSL0001:00``` as can be seen here:
http://i.imgur.com/dqvKaUt.jpg?1

I just need to figure out a way to mount the Ubuntu 15.10 Live USB rw instead of ro, so I can copy files and relevant output to that USB drive instead of taking pictures etc.
If you could supply me with some input regarding interesting commands to gather output from, or files you'd like to have, that would be something I can supply.

My list so far:
- ```ls /sys/bus/acpi/devices/```
- https://github.com/Manouchehri/vi8/issues/1#issuecomment-129013952

onitake, 2016-01-28:
Great!

I'm not sure if Ubuntu Live is persistent, so you may not be able to copy files back. But you can always just create a separate partition or use a memory card.

As mentioned, please provide the DSDT. It contains all the vital information about the hardware.

djvdorp, 2016-01-28:
@onitake great success! I managed to create a separate persistent partition on the usb drive to copy files back to, with the size of all the remaining free space. All commands from my list above have been executed and can be found here:
https://github.com/djvdorp/hi8/tree/master/resources/linux

onitake, 2016-01-28:
This is most interesting. The _DSM section is new, haven't seen that before. It looks like every vender fiddles something new into their ACPI tables? Anyway, the UUID in that DSM is mentioned [here](https://msdn.microsoft.com/en-us/library/windows/hardware/jj131711%28v=vs.85%29.aspx). Looks like it's used to get a HID descriptor address? Hmm... Perhaps there is a way to ask the controller for information after all.

Anyway, the MSSL0001 _HID seems odd, but the _CID matches what we have in other devices. I'm not sure if the ACPI functionality in the Linux kernel automatically matches both _HIDs and _CIDs, but if it does, no modifications should be required.

The rest of the descriptor is more or less what's known. I2C5 in the dependencies (still don't know why), device on I2C controller 4, I2C address 0x40, I2C speed 400kHz, interrupt pin 0x44, shutdown pin 0x1A on GPIO block 1, power save methods defined.

So, if the driver matches the _CID, it should work out of the box. You already have the firmware, so just copy that to /lib/firmware and you're set.

Note: You may have to find out what the exact dimensions of the touchscreen are, how many multitouch points your device supports, and if your firmware has finger tracking. The fwtool command line in the README is just an example. So, prepare for some try&error.

Recommended test order to get a proper firmware:
* Display dimensions 4095x4095, 10 multitouch points, kernel finger tracking off
* See if the driver works at all, test with `xinput test <dev number>`
* Determine the real dimensions of the touchscreen, see if you get multitouch events if you put all fingers on the screen
* Change the dimensions so they are a little bit above the maximum range you got, enable kernel finger tracking if necessary
* If finger tracking was already working, change the number of touch points to match (you don't have to do this if you use kernel finger tracking)
* Once you have a firmware with proper parameters, only minimal calibration with xinput_calibrator will be required.

djvdorp, 2016-01-28:
@onitake A little bit offtopic, but to get further I need to get myself a persistent Ubuntu installation on the Chuwi Hi8 tablet I think. For example, if I want to install wifi or bluetooth drivers I need kernel customizations that will be effective after a reboot only. Do you have any tips or experience in this matter? And if I would go on to delete Android and/or Windows10 from this device, how hard would it be to jump back to a factory image if I screw up?

So I went on Googling and it seems we are not the only ones working towards this goal (linux on atom based devices such as this Chuwi Hi8):

- http://techtablets.com/forum/topic/linux-on-atom-based-devices/
- http://techtablets.com/forum/topic/linux-mint-on-a-chuwi-hi10-tablet/
- https://www.reddit.com/r/linuxmasterrace/comments/428cap/xpost_rdebian_after_long_hours_i_successfully/czb5klm
- http://forum.xda-developers.com/windows-8-rt/general/chuwi-vi10-dual-boot-win-8-1-android-4-4-t3112954

onitake, 2016-01-28:
Ok, if you have a preinstalled OS, I recommend you pull a full dump to a flash stick or SD card. As long as you can still boot Linux, you can easily just dd that disk image back to the internal flash.

This is (roughly) what I did on my Trekstor: `dd if=/dev/mmcblk0 bs=1M | xz -c | dd bs=1M of=/media/user/stick/trekstor.img.gz`

The internal flash is usually not that large on these devices, so a xzipped image will not take up more than a few GB.

onitake, 2016-01-28:
As for the customised kernel: The first step I'd take is just getting the drivers to work. Compile them as modules out-of-tree, install them, test. When everything works, you can either turn the drivers into DKMS packages or hard-lock the kernel version so it doesn't get updated.

The ultimate goal will (obviously) be to get all the required modules into mainline Linux, but we are not there yet. rtl8723bs (the WiFi chip most commonly used in Baytrail tablets) and silead are getting there, but it will take time. Robert Dolca has submitted his silead_ts driver to linux-input, but he's not providing any support and testing on existing devices, so I'm currently more inclined to suggest that people should use gslx680_acpi instead. As long as nobody pulls hardware manufacturers back into the boat, silead_ts is a poor choice, even if the driver code is much better.

djvdorp, 2016-01-29:
@onitake, @barni2000 Update from my side: I have made a full dump of ```mmcblk0``` to my MicroSD card ```mmcblk1```, and have decided to go all-in into development. This means that at the next opportunity I have, I will remove all partitions on the device and install Ubuntu 15.10 on it for now. 

However, there is something weird that does not compute for me right now; [the BIOS](https://github.com/djvdorp/hi8/blob/master/miscinfo.md) for the Chuwi Hi8 claims that the Wifi (and Bluetooth) are supplied by a ```Realtek RGN RTL8723BS```. However, if I check my [dmesg output](https://github.com/djvdorp/hi8/blob/master/resources/linux/dmesg.txt), it has some lines that seem to indicate otherwise:
```
[  395.962397] brcmf_sdio_drivestrengthinit: No SDIO Drive strength init done for chip 43430 rev 0 pmurev 24
[  395.963219] usbcore: registered new interface driver brcmfmac
[  395.964186] brcmfmac_sdio mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.bin failed with error -2
[  396.969792] brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[  397.977803] brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```

This seems to correspond to a ```Broadcom BCM43430``` wireless chipset which it is trying to use via ```brcmfmac```. Also, the [windows drivers](https://github.com/djvdorp/hi8/tree/master/drivers/windows/wireless) seem to indicate so. How can I be sure/more certain which wireless chip is powering my device, so I know what drivers to install and use?

barni2000, 2016-01-29:
Chuwi Hi8 has broadcom wifi and realtek bluetooth.

djvdorp, 2016-01-29:
@barni2000 Thanks for the quick reply. Do you by any chance have an idea how to resolve the broadcom wifi error ```brcmfmac_sdio mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.bin failed with error -2``` ?

barni2000, 2016-01-29:
I have no idea, try the broadcom-wl driver.

onitake, 2016-01-31:
This firmware is in the [linux-firmware tree](http://git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git/tree/brcm). Debian/Ubuntu may also have it in their firmware-linux-nonfree packages. Not sure.

wpa2, 2016-03-19:
I would like to revive the topic. I tried to get to work touchscreen on my Сhuwi hi8 - without result
.

Run from a USB stick lubuntu-14.10-desktop-amd64
in gslx680-acpi:
make
```
lubuntu@lubuntu:/media/lubuntu/8765-4321/drv/gslx680-acpi$ make
make -C /lib/modules/3.16.0-23-generic/build ARCH=x86_64 CROSS_COMPILE= M=/media/lubuntu/8765-4321/drv/gslx680-acpi modules
make[1]: Entering directory '/usr/src/linux-headers-3.16.0-23-generic'
  CC [M]  /media/lubuntu/8765-4321/drv/gslx680-acpi/gslx680_ts_acpi.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /media/lubuntu/8765-4321/drv/gslx680-acpi/gslx680_ts_acpi.mod.o
  LD [M]  /media/lubuntu/8765-4321/drv/gslx680-acpi/gslx680_ts_acpi.ko
make[1]: Leaving directory '/usr/src/linux-headers-3.16.0-23-generic'
```
sudo make install
```
lubuntu@lubuntu:/media/lubuntu/8765-4321/drv/gslx680-acpi$ sudo make install
install -p -m 644 gslx680_ts_acpi.ko  /lib/modules/3.16.0-23-generic/kernel/drivers/input/touchscreen
/sbin/depmod -a 3.16.0-23-generic
```
after first `sudo modprobe gslx680_ts_acpi` in dmesg appeared only:
`[  646.344152] gslx680_ts_acpi: module verification failed: signature and/or  required key missing - tainting kernel
`
next  insertion and removal of the module did not give messages to dmesg

in gsl-firmware
take firmware.fw from firmware/chuwi/hi8 and convert by:
`./fwtool -c firmware.fw -m 1680 -w 940 -h 750 -t 5 silead_ts.fw`
then copy silead_ts.fw to /lib/firmware/
-w and -h data should be absolutely correct, or with any should work somehow?

some more info:
```
lubuntu@lubuntu:~/gsl-firmware/tools$ modinfo gslx680_ts_acpi
filename:       /lib/modules/3.16.0-23-generic/kernel/drivers/input/touchscreen/gslx680_ts_acpi.ko
license:        GPL
version:        2.0.0
author:         Gregor Riepl <onitake@gmail.com>
description:    GSLX680 touchscreen controller driver
srcversion:     144899E1944F61F1A591AA0
alias:          i2c:gslx680
alias:          acpi*:PNP3680:*
alias:          acpi*:PNP1680:*
alias:          acpi*:MSSL3680:*
alias:          acpi*:MSSL1680:*
depends:        
vermagic:       3.16.0-23-generic SMP mod_unload modversions 

```
```
lubuntu@lubuntu:~$ ls /sys/bus/acpi/devices/MSSL
MSSL0001:00/ MSSL1680:00/ 

```
[
[DSDT_hi8.txt](https://github.com/onitake/gslx680-acpi/files/181079/DSDT_hi8.txt)
](url)
in DSDT I find 2 parts with "1680"
```
Device (TCS0)
            {
                Name (_ADR, Zero)  // _ADR: Address
                Name (_HID, "MSSL1680")  // _HID: Hardware ID
                Name (_CID, "PNP1680")  // _CID: Compatible ID
                Name (_HRV, One)  // _HRV: Hardware Revision
                Name (_S0W, Zero)  // _S0W: S0 Device Wake State
                Name (_DEP, Package (0x03)  // _DEP: Dependencies
                {
                    GPO1, 
                    I2C5, 
                    ^^I2C5.PMI1
                })
```
```
Device (TCS1)
            {
                Name (_ADR, Zero)  // _ADR: Address
                Name (_HID, "MSSL0001")  // _HID: Hardware ID
                Name (_CID, "PNP1680")  // _CID: Compatible ID
                Name (_HRV, One)  // _HRV: Hardware Revision
                Name (_S0W, Zero)  // _S0W: S0 Device Wake State
                Name (_DEP, Package (0x03)  // _DEP: Dependencies
                {
                    GPO1, 
                    I2C5, 
                    ^^I2C5.PMI1
                })
```

What else should/can I do?

djvdorp, 2016-03-20:
@wpa2 Unfortunately I can not help you with this anymore, since I sold my Hi8, sorry.

wpa2, 2016-03-20:
@djvdorp , so you don't get touchscreen to work?

@onitake , maybe you are interested in continuing with this tablet? if so, say what to do, what other information is needed, I will try to do everything.

onitake, 2016-03-20:
You can ignore the signature verification message. That happens because Ubuntu kernels are signed, and out-of-tree built modules are not. They will still load (unless you enable the full secure boot chain of trust).

Can you grep the dmesg output for messages containing `MSSL`?

Also, please post the output of `xinput list`, if possible.

wpa2, 2016-03-20:
@onitake 

```
lubuntu@lubuntu:~$ xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ SIGMACHIP USB Keyboard                  	id=7	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Video Bus                               	id=6	[slave  keyboard (3)]
    ↳ gpio-keys                               	id=9	[slave  keyboard (3)]
    ↳ gpio-keys                               	id=10	[slave  keyboard (3)]
    ↳ SIGMACHIP USB Keyboard                  	id=8	[slave  keyboard (3)]

```
Nothing about MSSL in dmesg:
```
lubuntu@lubuntu:~$ dmesg | grep MSSL
lubuntu@lubuntu:~$ dmesg | grep -i MSSL
lubuntu@lubuntu:~$ dmesg | grep -i MSS
lubuntu@lubuntu:~$ 

```

onitake, 2016-03-20:
Umm... If you don't get any messages containing "MSSL", that probably means the driver didn't load at all, or it didn't match any hardware.
Can you skim the kernel log/dmesg some more and see if you find anything? The driver should at least print something like `gsl_ts_probe: got a device named MSSL1680 at address 0x40, IRQ 201, flags 0x0`

wpa2, 2016-03-21:
I made all steps from beginning and now find that in dmesg:
```
[ 1200.040258] gslx680_ts_acpi: module verification failed: signature and/or  required key missing - tainting kernel
[ 1200.041519] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ 68, flags 0x0
[ 1200.041671] ACPI Exception: AE_ERROR, Returned by Handler for [GeneralPurposeIo] (20140424/evregion-297)
[ 1200.041680] ACPI Error: Method parse/execution failed [\_SB_.I2C4.TCS0._PS0] (Node ffff88007b437898), AE_ERROR (20140424/psparse-536)
[ 1200.041693] acpi MSSL1680:00: Failed to change power state to D0
[ 1200.041698] gslx680 i2c-MSSL1680:00: gsl_ts_probe: failed to wake up device through ACPI: -19, continuting anyway
[ 1200.042068] input: Silead GSLx680 Touchscreen as /devices/platform/80860F41:03/i2c-11/i2c-MSSL1680:00/input/input8
[ 1200.063372] gslx680 i2c-MSSL1680:00: gsl_ts_reset_chip: gsl_ts_write 1 fail!
[ 1200.063381] gslx680 i2c-MSSL1680:00: gsl_ts_probe: chip reset failed
[ 1200.095585] gslx680: probe of i2c-MSSL1680:00 failed with error -121
```
I hope this can help...  ?

onitake, 2016-03-21:
Thanks.

Yes, this is a known problem that started occurring on Ubuntu kernel 4.2.0-27 and later on my test device. Kernel 4.2.0-25 still worked fine. You can try downgrading and see if it helps. Also, please pull the latest sources from github. I changed a few things to make the driver more reliable.

Once I know why this happens, I'll send a bug report to Launchpad and the kernel developers. I suspect it's a BIOS bug that didn't trigger before 4.2.0-25. If you can find a BIOS upgrade, please try that as well.

wpa2, 2016-03-22:
@onitake 
Sorry for delay. Long time picking up the appropriate settings.
My touchscreen has started to work! Thank you for your work and support!
Now that is what I got:
My kernel older - 3.16.0-23
```
lubuntu@lubuntu:~$ uname -a
Linux lubuntu 3.16.0-23-generic #31-Ubuntu SMP Tue Oct 21 17:56:17 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
```
I will try other versions of Linux later.

I download new version of the module and it worked.
```
[ 2687.665808] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ 68, flags 0x0
[ 2687.666013] input: Silead GSLx680 Touchscreen as /devices/platform/80860F41:03/i2c-11/i2c-MSSL1680:00/input/input14
[ 2689.938792] byt-rt5640 byt-rt5640: ASoC: CPU DAI baytrail-pcm-audio not registered
[ 2689.938825] platform byt-rt5640: Driver byt-rt5640 requests probe deferral

```
To convert firmware i used command:
`
./fwtool -c firmware.fw -m 1680 -w 1140 -h 1650 -t 10 -f swap silead_ts.fw
`
if necessary, I can upload the file silesd.fw later - I forgot to save it to flash drive.

wpa2, 2016-03-24:
[silead_ts.fw.zip](https://github.com/onitake/gslx680-acpi/files/187808/silead_ts.fw.zip)
delete ".zip" before use.

onitake, 2016-03-24:
Can you create a pull request on https://github.com/onitake/gsl-firmware for the updated Hi8 firmware?

Please use https://github.com/onitake/gsl-firmware/blob/master/firmware/trekstor/surftab7old/README.md as a template for your README.md. `Finger tracking` should be `Yes` in your case, as you didn't set the `track` flag, and `Axes swapped` should be `Yes` too.

wpa2, 2016-03-25:
@onitake 
Done.
I did it for the first time, so something may be wrong. Check, please.

onitake, 2016-03-26:
Github messed up the paths, but I already corrected that.
Thank you!

youling257, 2016-03-27:
@wpa2 
Chuwi/hi8 ，8 inch，Display resolution	1920x1200 ，Touch panel resolution，1665x1140
1665x1140，how do you know ? do you try it on Androidx86?

wpa2, 2016-03-27:
@youling257 
No. I try it in Ubuntu 14.10. I picked it experimentally
maybe it is not absolutely precisely, but very close.

youling257, 2016-03-27:
http://www.fosshub.com/Android-x86.html ，android-x86-6.0_20160318，has gslx680_ts_acpi ，extract system.sfs，mount system.img，cp your silead_ts.fw to /lib/firmware，then boot Androidx86，Android Terminal，modprobe -r gslx680_ts_acpi，modprobe gslx680_ts_acpi

onitake, 2016-03-27:
Is it really necessary to do
```
modprobe -r gslx680_ts_acpi
modprobe gslx680_ts_acpi
```
?

If you have silead_ts.fw in the right place, it should load automatically at boot and Android should work right out of the box.

m2marliave, 2016-06-04:
I know this post is old but could someone help me installing a ubuntu based distribution on the Chuwi Hi8 ? I'm a bit noob but I would like to make touch work

matthew11, 2020-10-12:
Hi!

I'm trying to install the latest Ubuntu on my Chuwi HI8. I managed to install it, fixed the bootloader, and the WIFI, but I'm stuck with the touch screen.

I managed to compile the driver, and also inserted into the kernel, however it looks like it doesn't match for the driver. (got nothing in dmesg apart from tainting the kernel with an unsigned driver)
`[ 1046.361041] gslx680_ts_acpi: module verification failed: signature and/or required key missing - tainting kernel`
I had driver issues on Windows as well (after a failed update) this is why I replaced with the Ubuntu. It looks like the Chuwi HI8 have many variant, I have the model CWI509.

Could you please help me to debug the situation? Reading the comments I suppose the DSDT will provide helpful information, so I attached this [file.]
[dsdt.zip](https://github.com/onitake/gslx680-acpi/files/5367582/dsdt.zip)

Thank you in advance!

closed 2020-10-13 by onitake

matthew11, 2020-10-14:
I have some major breakthrough. The problem was not with the driver, but somehow the hardware (bus?) not initialized properly.
I have booted into the factory-installed Android which working properly (just very outdated), and after a reboot the touchscreen was started to work properly (using your driver) on Ubuntu as well. As a side effect the accelerometer, and the Bluetooth is also started to work, moreover I have the manufacturer's logo on Ubuntu's boot animation (LOL).

I do not intend to go off-topic on this thread (which is already closed), but I think this useful information is worth sharing for others might facing similar issue.

onitake, 2020-10-14:
That really sounds like an initialization issue.
The Android probably sent the right startup sequence or firmware to make the peripherals run.
I wonder what happens if you put it to sleep. Chances are it won't come back properly.

matthew11, 2020-10-16:
Yes, that was what I assumed, but surprisingly the device stays initialized not only after a resume, but after full power cycles. I do even completely depleted the battery to see what happens, and it is still working fine. To be more precise I cannot reproduce this uninitialized state now, not even with long-pressing the power button. I start to believe it is not an initialization issue - at least not in a common sense - but a configuration one. Somehow the Android set up the firmware's configuration properly. I bet the devices would work on Windows as well now. Maybe the Windows (driver) update was what messed up this configuration in the first place. Strange indeed.