onitake/gslx680-acpi #5: Wiki with support matrix.

barni2000, 2016-01-26:
I think we should create a wiki with some informations for tablets.

For example:
SupportMatrix:

| Device     | Tested | Old stlye frimware          | New style firmware          | Comment         |
|------------|--------|-----------------------------|-----------------------------|-----------------|
| Utok i800  |  yes | Link for old style firmware | Link for new style firmware | Something       |
| Chuwi hi8  |  no | Link for old style firmware | Link for new style firmware | Not working yet |
| Chuwi vi8  |  no | Link for old style firmware | Link for new style firmware | Should test     |

onitake, 2016-01-26:
That's a great idea, though I was considering moving the firmware conversion tools and all the firmware images into a separate repository.

barni2000, 2016-01-29:
Could you add the wiki function for this repository?

onitake, 2016-02-02:
I'll set up a new repo that should make it much easier to share firmware images for various devices.
I want to reduce the "pollution" of the driver source code as much as possible.

onitake, 2016-02-02:
The new repo is here: https://github.com/onitake/gsl-firmware

I'm moving all the tools and stuff I've collected so far there. If you'd like to add anything, send a pull request or a link. Do you think a Wiki is still necessary? I prefer to keep the documentation in files instead, so they can be shared and edited locally more easily.

closed 2016-02-02 by barni2000