onitake/gslx680-acpi #6: Right click?

sigboe, 2016-01-29:
How do I configure right click to work, either two finger or long press is sufficient for me.

I tried adding these lines 
```
Option "EmulateThirdButton" "1"
Option "EmulateThirdButtonTimeout" "750"
Option "EmulateThirdButtonMoveThreshold" "30"
```

into the same section statement as the calibration data from `xinput_calibrator ` but longpress does nothing.

onitake, 2016-01-31:
There are multiple reasons why this does not work right. Touchscreen input has a lot of jitter and I'm not doing anything to reduce it in the driver. So, a "long" click may instead be recognized as a drag or multiple clicks. I'll see if finger tracking can be tuned a little.

I've had multitouch working occasionally, which may also be used to emulate second and third button clicks.

sigboe, 2016-02-01:
I would be happy with two finger click! Or long press, anything really.

sigboe, 2016-02-02:
Might having the correct resolution in the firmware, and not needing xinput calibration help with the situation?

onitake, 2016-02-04:
Good question. I think it will surely help with touch detection close to the edges, but jitter also affects finger tracking, and that happens before scaling. Guess I need to check how other drivers do it.

In any case, finding out the correct dimensions is still a good idea.