onitake/gslx680-acpi #7: Setting axis in the firmware?

sigboe, 2016-02-03:
So I think I figured out the resolution of my touch screen to be 1700x1125. Or at least if I start with a fresh boot with display settings clockwise, load a firmware with -w 1700 and -h 1125 it tracks 1:1 but one of the axis is reversed. Is there a way to specify this in the firmware?

onitake, 2016-02-04:
Yes, in fact there is. Use fwtool to set the xflip or yflip flag. The feature is still untested, though.

sigboe, 2016-02-04:
That's great, gonna try this today. Might I suggest you rename them to `-wflip` and `-hflip` to be consistent with `-w` and `-h`

It's sad that the calibration only works for the orientation it was made for :disappointed: but that looks to be a problem with X, the relic that X is.

onitake, 2016-02-04:
Ok. Sounds good, but I'm just going to change that in fwtool, not the driver.

If the calibration is wrong, it's not that big a deal to redo it, I think. :)
Optimally, one shouldn't have to do any calibration at all.

onitake, 2016-02-04:
On second thought, hflip is ambiguous. It could be interpreted as 'horizontal' flip.

I'll keep the parameters as they are, perhaps with some clarifications in the help text and readme. I could turn the various -f arguments into standalone parameters, but one-letter options are better IMHO.

commit ba60173b (2016-02-04): Added clarification about axis mirroring and other flags.
Closes #7.
closed 2016-02-04 by onitake