onitake/gslx680-acpi #8: CCE TF74W support

tiagosales, 2016-02-06:
Hello,

I have a brazilian tablet CCE that have Windows as only OS and Silead driver on my device using android x86 doesn't work.

I compiled the driver using android x86 source kernel 4.0.9, did a insmod and the error below is showed.

```
<4>[  467.467528] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ 200, flags 0x0
<6>[  467.476087] input: Silead GSLx680 Touchscreen as /devices/platform/80860F41:01/i2c-1/i2c-MSSL1680:00/input/input10
<3>[  467.488589] gslx680 i2c-MSSL1680:00: gsl_ts_reset_chip: gsl_ts_write 1 fail!
<3>[  467.488612] gslx680 i2c-MSSL1680:00: gsl_ts_probe: chip reset failed
<4>[  467.516654] gslx680: probe of i2c-MSSL1680:00 failed with error -121
```

I extracted the FW from SileadTouch.sys(Windows driver) using a hex editor, and placed the .fw converted by fwtool on /lib/firmware folder.

Attached the DSDT.dsl.
[dsdt.dsl.txt](https://github.com/onitake/gslx680-acpi/files/120474/dsdt.dsl.txt)

Can you help me?

onitake, 2016-02-07:
You seem to have entries for both Goodix and Silead ts controllers in your DSDT.
Did you verify that your device really contains a Silead?

Anyway, the error happens during the very first I2C transfer. Errno 121 is EREMOTEIO. This indicates that nothing is responding on the I2C bus, which could either mean that the device is not awake or that there's simply nothing there.

Have you tried the rastersoft driver? Perhaps it's more successful...

And on a completely different note: Can you open an issue in https://github.com/onitake/gsl-firmware with information about your device and a link to the Windows drivers?

tiagosales, 2016-02-08:
Hello,

The Rastersoft driver works good, I need test some dimensions to be usable at least , but this works. Not necessary to inform the gpio pin..

I will add the driver in the repo. 

Thank you for attention and help.

onitake, 2016-03-24:
Thanks, I'm closing this issue now.
Please reopen if you still need help.

closed 2016-03-24 by onitake