onitake/gslx680-acpi #9: gsl_ts_reset_chip: gsl_ts_write 1 fail!

barni2000, 2016-03-19:
I have got some new errors:
[  585.405027] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ 201, flags 0x0
[  585.477655] input: Silead GSLx680 Touchscreen as /devices/platform/80860F41:03/i2c-11/i2c-MSSL1680:00/input/input10
[  585.504702] gslx680 i2c-MSSL1680:00: gsl_ts_reset_chip: gsl_ts_write 1 fail!
[  585.504715] gslx680 i2c-MSSL1680:00: gsl_ts_probe: chip reset failed
[  585.544934] gslx680: probe of i2c-MSSL1680:00 failed with error -121

onitake, 2016-03-19:
Yeah, sorry about that. This happens when the chip has not been properly woken up.

I removed the ACPI wakeup procedure and replaced it with GPIO signalling, as I ran into a conflict on my tablet. But there's still something wrong. I'll update you when I get it working again.

closed 2016-03-20 by barni2000

onitake, 2016-03-20:
Did you close the issue because it's working again or because you can't find a solution?

After some experimentation, I found that the Silead in my tablet does not wake up properly when I just signal the GPIO line. So I added the ACPI method call back in, which manipulates two GPIO pins in a certain sequence. If that doesn't work or ACPI is not available, it uses the GPIO pin instead.

This make wakeup work again here, but there seems to be a regression in Ubuntu kernel 4.2.0-27, causing it to fail. 4.2.0-25 still works. It looks like a broken DSDT that triggers errors in newer kernels. Too bad there's no BIOS updates for these cheap tablets.