onitake/gslx680-acpi #11: Driver not working on 4.2.0-27 and later

onitake, 2016-03-24:
It seems that gslx680-acpi has stopped working on Ubuntu kernel 4.2.0-27, at least for some devices.

The ACPI power method wakeup procedure fails due to either an error in the kernel's DSDT parsing code or the Designware I2C driver. The exact cause has not been Identified yet.

nerijus, 2017-12-02:
Is it because of this it does not work on Fedora 27 kernel 4.13.15-300.fc27.x86_64?
dmesg:
```
[  499.464187] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ 183, flags 0x0
[  499.469711] input: Silead GSLx680 Touchscreen as /devices/pci0000:00/808622C1:05/i2c-5/i2c-MSSL1680:00/input/input11
[  499.473056] gslx680 i2c-MSSL1680:00: gsl_ts_probe: failed to wake up device through ACPI: 0, using GPIO controls instead
[  499.473070] gslx680 i2c-MSSL1680:00: gsl_ts_probe: error obtaining power pin GPIO resource
[  499.488246] gslx680: probe of i2c-MSSL1680:00 failed with error -2
```

onitake, 2017-12-03:
Yes, that is most likely the cause. I stopped working on gslx680-acpi since silead.ko made it into the kernel.
Not sure if it will fix wakeup for your device, but please try.