onitake/gslx680-acpi #13: Cube iwork8 Ultimate support

NotKit, 2016-03-29:
Hello. 
I'm trying to get touchscreen working on Cube iwork8 Ultimate. It's an Cherry Trail-based dual boot tablet. Unfortunately, loading driver on 4.4.1 fails, probably due to issue #11.
```
[378.358143] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ 186, flags 0x0
[378.358404] gslx680 i2c-MSSL1680:00: gsl_ts_probe: error obtaining power pin GPIO resource
[378.522120] gslx680: probe of i2c-MSSL1680:00 failed with error -16
```
Stock Android GSL1680 driver is compiled into kernel. I loaded it in disassembler, seems to contain multiple firmwares, along with detection code (gsl_identify_tp at FFFFFFFF816F2BD0). From what I can understand, it first loads generic firmware, gets touchscreen type and then loads firmware for the device. For my device tp_type is 27, judging from dmesg, so I extracted the firmware that gets loaded for this type (at FFFFFFFF81AD5F20) manually, but not completely sure it's correct.

I also tried using gslx680-acpi driver with Android-IA Cherry Trail kernel 3.14.37 (it's for Intel reference boards and lacks tablet-specific adaptations, but support for tablet's hardware is still better than with mainline kernel). For some reason it doesn't set IRQ. I wonder why can this happen:
```
[157.879029] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ -1, flags 0x0
[157.879042] gslx680 i2c-MSSL1680:00: gsl_ts_probe: missing IRQ configuration
```

[cube_iwork8_ultimate.tar.gz](https://github.com/onitake/gslx680-acpi/files/193513/cube_iwork8_ultimate.tar.gz)

onitake, 2016-03-29:
Did you try fw_extractor from https://github.com/onitake/gsl-firmware/tree/master/tools ?
It should be able to extract all firmware blobs from the Android driver.

A missing IRQ configuration can only happen if the ACPI DSDT record for the device lacks the information. Can you post a dump of your DSDT or the decompiled section for MSSL1680 here?
See here for instructions: https://wiki.archlinux.org/index.php/DSDT#Recompiling_it_yourself

NotKit, 2016-03-30:
I've included the relevant information (DSDT, dmesg logs, firmware) in cube_iwork8_ultimate.tar.gz in post above. fw_extractor wouldn't work, as it relies on readelf, which works for modules, but here driver and firmwares are compiled into kernel and it lacks symbols.

On mainline Linux 4.4 kernel the driver works if I comment out power management. I converted firmware with the following command:
`./fwtool -c firmware.fw -w 880 -h 1660 -t 10 -f swap silead_ts.fw`

[DSDT.zip](https://github.com/onitake/gslx680-acpi/files/194607/DSDT.zip)