onitake/gslx680-acpi #15: CHUWI HI10

zak53, 2016-06-04:
HI
I have CHUWI HI10 and i just want to know if this driver work with this tablet please.

onitake, 2016-06-04:
Yes, it should work.
Use the firmware here: https://github.com/onitake/gsl-firmware/tree/master/firmware/chuwi/vi10 (you need silead_ts.fw)

onitake, 2016-06-04:
Ah wait, sorry. This is for the vi10, not the hi10.
I don't know if it works, but you definitely need firmware. Search for the Windows/Android drivers and provide a link, and I may be able to help you.

zak53, 2016-06-04:
Hi 
Thank you,
Here is the .zip you can find all drivers for hi10, but if you tell me where i can find the firmware i can search,
My ubuntu can't boot on graphical mode only with (nomodeset) option on the grub i don't know to fix this issue.
Thank you