onitake/gslx680-acpi #16: cube i10w

kahrloz, 2016-07-05:
i need help with the fw files that i pulled from my tablet's.ko, i need to know how do i use these in the Makefile?

onitake, 2016-07-05:
You need to find out which one suits your tablet first.
Then, you have to figure out the correct device parameters.
If you have those, use fwtool to convert the firmware,  then copy it to /lib/firmware.
See [the instructions](https://github.com/onitake/gsl-firmware/blob/master/README.md#gslx680-acpi) on how to do this.

If you don't know the right firmware or parameters yet, you have to experiment. Test each firmware file and choose the one that works best. For the parameters, it may help to look at a similar device.

kahrloz, 2016-07-05:
OK so I got it to work but touch is slightly inaccurate, even after start X

kahrloz, 2016-07-05:
I mean after xinput calibrator

onitake, 2016-07-06:
Ok, first hurdle taken.
The rest is purely firmware related, so it makes sense to move further discussion to [gsl-firmware#20](https://github.com/onitake/gsl-firmware/issues/20).
I'm closing this bug report now.

closed 2016-07-06 by onitake

kahrloz, 2016-07-06:
./fwtool -c  firmware.fw -m 1680 -w 1366 -h 768 -t 10 silead_ts.fw

kahrloz, 2016-07-06:
It is working perfectly now, I ran the xinput calibrator like six times

kahrloz, 2016-07-06:
Now I'm off into enabling Bluetooth

onitake, 2016-07-06:
Can you post a link to the Android driver you used, and tell me which firmware file you had success with exactly?
Or, better yet, submit a pull request according to [these guidelines](https://github.com/onitake/gsl-firmware#adding-new-firmware).

kahrloz, 2016-07-06:
I ended up using the windows drivers. The Android one only had like half of
the touchscreen unresponsive.

kahrloz, 2016-07-07:
https://drive.google.com/open?id=0B3eEskvA0DSXZ0RZRVJnR2t1RE0
that’s the drivers i ended up using from windows

onitake, 2016-07-07:
Is that the "official" link where you can get those drivers, or is there a vendor website with a driver package too?

kahrloz, 2016-07-08:
The drivers are hosted at techtablets.com