onitake/gslx680-acpi #18: Android-x86 / RemixOS calibration fix

arjanvlek, 2016-08-09:
Hi,

Android X86 / RemixOS work on tablets with this touchscreen and it does register touches, but the calibration is off. How can the screen properly be calibrated???

onitake, 2016-08-09:
I don't use gslx680-acpi with Android myself, so I can't help much here.

There is one critical point that you should tune, and that is the touch screen resolution. If you get that right, you should be able to calibrate it with the tools built into Android itself.

There is a lenghty issue thread from another AndroidX86 user [https://github.com/onitake/gsl-firmware/issues/6](here) - perhaps it can give you some hints?

arjanvlek, 2016-08-10:
I found this firmware for my tablet: https://github.com/onitake/gsl-firmware/blob/master/firmware/trekstor/surftab7new/silead_ts.fw

In which folder of the Android system should I place this file?

arjanvlek, 2016-08-10:
Yes! Got it to work!!!
It was pretty simple, as android-x86 already uses gslx680-acpi.
All what had to be done was to replace the file /lib/firmware/silead_ts.fw with this one:
https://github.com/onitake/gsl-firmware/blob/master/firmware/trekstor/surftab7new/silead_ts.fw

Commands:
`
su
cd /lib/firmware
rm silead_ts.fw
cp /storage/emulated/0/Download/silead_ts.fw .
chmod 644 silead_ts.fw
reboot
`

closed 2016-08-10 by arjanvlek

onitake, 2016-08-11:
Great!
Have fun! :)

devsed, 2019-01-28:
arjanvlek, I get "read-only filesystem", when trying to do rm and cp and can't chmod etc. How should this be used? (with real device)

pekezu, 2019-01-28:
arjanvlek, I get "read-only filesystem", when trying to do rm and cp and can't chmod etc. How should this be used? (with real device)