onitake/gslx680-acpi #19: gslx680 i2c-MSSL1680:00: gsl_ts_reset_chip: gsl_ts_write 1 fail!

youling257, 2016-11-24:
[   14.497512] gslx680 i2c-MSSL1680:00: gsl_ts_reset_chip: gsl_ts_write 1 fail!
[   14.503443] gslx680 i2c-MSSL1680:00: gsl_ts_probe: chip reset failed
[   14.533874] gslx680: probe of i2c-MSSL1680:00 failed with error -11

onitake, 2016-11-24:
That is not a proper bug report.

Please write at least:
- When does it happen (always/sometimes/under certain conditions)
- Which device
- Which kernel and driver version
- Did it work before? Did you change anything?

youling257, 2016-11-24:
it work now on https://github.com/maurossi/linux/tree/kernel-4.8 

it doesn't work on https://osdn.net/projects/android-x86/scm/git/kernel/tree/kernel-4.9/

youling257, 2016-11-24:
Who has encountered the same problem ，How he solved it ？

onitake, 2016-11-24:
Hmmm...

I don't have time to work on the project at the moment, but I know that there is now a new Silead driver in kernel 4.8. It will not work correctly with Baytrail based devices, but I know that the maintainer wants to improve Intel support.

I suggest you stay on kernel 4.8 for the moment and watch closely for an update on the new driver. Once it's ready for general consumption, you should be able to get your touchscreen working without installing extra software.

youling257, 2016-11-24:
on 4.8 kernel，work ；on 4.9 ，doesn't work

youling257, 2016-11-26:
on https://github.com/maurossi/linux/tree/kernel-4.8 ，silead.ko、mssl1680.fw and gslx680_ts_acpi.ko、silead_ts.fw，all can work well .
but on https://osdn.net/projects/android-x86/scm/git/kernel/tree/kernel-4.9/ 

[   13.806770] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ 68, flags 0x0
[   13.914450] gslx680 i2c-MSSL1680:00: gsl_ts_reset_chip: gsl_ts_write 1 fail!
[   13.920399] gslx680 i2c-MSSL1680:00: gsl_ts_probe: chip reset failed
[   13.947868] gslx680: probe of i2c-MSSL1680:00 failed with error -11

root@x86:/ # dmesg | grep silead
[   14.031463] silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
[   14.075816] silead_ts i2c-MSSL1680:00: Registers clear error -11
[   14.087734] silead_ts: probe of i2c-MSSL1680:00 failed with error -11
[  180.627479] silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
[  180.628877] silead_ts i2c-MSSL1680:00: Registers clear error -11
[  180.638825] silead_ts: probe of i2c-MSSL1680:00 failed with error -11
root@x86:/ #

youling257, 2016-11-28:
root@youling257-ONDA-Tablet:/home/youling257# dmesg | grep silead
[   10.889503] silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
[   10.890766] silead_ts i2c-MSSL1680:00: Registers clear error -11
[   10.904226] silead_ts: probe of i2c-MSSL1680:00 failed with error -11
root@youling257-ONDA-Tablet:/home/youling257# dmesg | grep gslx
[  158.714454] gslx680_ts_acpi: loading out-of-tree module taints kernel.
[  158.714528] gslx680_ts_acpi: module verification failed: signature and/or required key missing - tainting kernel
[  158.737912] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ 197, flags 0x0
[  158.784426] gslx680 i2c-MSSL1680:00: gsl_ts_reset_chip: gsl_ts_write 1 fail!
[  158.784438] gslx680 i2c-MSSL1680:00: gsl_ts_probe: chip reset failed
[  158.838300] gslx680: probe of i2c-MSSL1680:00 failed with error -11
root@youling257-ONDA-Tablet:/home/youling257# uname -a
Linux youling257-ONDA-Tablet 4.9.0-1-generic #2-Ubuntu SMP Mon Nov 14 21:43:10 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
root@youling257-ONDA-Tablet:/home/youling257#

youling257, 2017-04-04:
no other people care it，closed .

closed 2017-04-04 by youling257

onitake, 2017-04-10:
I'm sorry about that, but there's not much that can be done at the moment.
I know that the linux-input developers are working on a fix, however.

The root of the problem is the configuration of the power control pin of the chip. Many tablets have conflicting DSDT data and the driver needs to work around that.

youling257, 2017-05-31:
4.12 kernel Fixed ，gslx680 work well on 4.12 kernel ，silead also work.

onitake, 2017-05-31:
Great to hear!
Thanks for testing and reporting back.

youling257, 2017-06-03:
I test 4.11.3 kernel，also working！

youling257, 2017-06-28:
Error -11 happen on 4.12 rc7 kernel !
gslx680 and silead work on 4.9.34 kernel，I used 4.9 kernel a month，4.9.30、31、32、34，gslx680 working.

rootd, 2018-09-16:
Any news on this? I haven't been able to get the touchscreen of my Teclast X98 II Plus working on any newer kernel (>= 4.13). I'm getting the same Registers clear error -11 error than described above

onitake, 2018-09-16:
I stopped working on gslx680-acpi some time ago, please try silead_ts.
If it doesn't work for you, please post a bug report on the linux-input mailing list: http://vger.kernel.org/vger-lists.html#linux-input

frakswe, 2018-09-20:
this might not be the right place but i need some clarity about this, are we still supposed to download the corresponding firmware from "gls-firmware/firmware" and put that in /lib/firmware? there seem to be a builtin kernel module atleast, it just doesn't seem to load firmware correctly. have i got that right?

onitake, 2018-09-20:
@frakswe Yes, this is the correct way.

The kernel driver will load the correct firmware if it can identify your device.
If it cannot, it will try to load firmware from a default location (/lib/firmware/silead/mss1680.fw I think), that will not work correctly almost 100% of the time. But in doing so, it will indicate in the kernel log that it couldn't find your device in the list of tested and supported devices. In that case, you will need to add support to the kernel driver yourself. The README gives a rough guide on how to accomplish this.

frakswe, 2018-09-20:
okay, atleast i know that i'm on the right track now. many thanks.