onitake/gslx680-acpi #20: 3/4 dead zone on CW1529 (Chuwi Hi10 Pro)

ssinyagin, 2017-03-14:
on my CW1529 (Chuwi Hi10 Pro), I could not make the touchscreen work properly. maximum that I could achieve is that the touchscreen works in the lower 1/4 part of the screen, but in the rest of the area the cursor is jumping randomly. Also the driver does not take the screen orientation into consideration.

So, for "normal" orientation, which is vertical by default, the best result is achieved with
```
perl -I tools/ ./tools/fwtool -c firmware/chuwi/hi10_pro/firmware.fw -m 1680 -w 1260 -h 1925 -t 5 -f track,swap /lib/firmware/silead_ts.fw
```

If I do `xrandr -o right`, and the screen is usable with the docked keyboard, the firmware needs different parameters:
```
perl -I tools/ ./tools/fwtool -c firmware/chuwi/hi10_pro/firmware.fw -m 1680 -w 1925 -h 1260 -t 5 -f track,yflip /lib/firmware/silead_ts.fw
```

I've got the original MMC image. Should I look for a newer firmware on it? Where approximately?

ssinyagin, 2017-03-14:
more details about building it: https://txlab.wordpress.com/2017/03/11/running-ubuntu-on-chuwi-hi10-pro-tablet/

onitake, 2017-03-14:
Nice, thanks for the writeup on your blog!

The erratic behaviour is almost definitely caused by incompatible firmware.
As I mentioned previously on other bug reports on gsl-firmware, the firmware image contains a set of device calibration parameters that are specific to the touch panel on a particular device.

If Chuwi has changed the panel in some hardware revision, new firmware is needed.
Can you obtain updated Windows drivers for this tablet and extract the firmware with [this tool here](https://github.com/angellsl10/I1D6-C109S-Hi10Pro/blob/master/touchscreen/sys2fw.sh)?
Maybe that will help.

hopkinskong, 2017-03-14:
Also try different firmware from the driver too (if there is).

youling257, 2017-03-14:
https://bugzilla.kernel.org/attachment.cgi?id=246851
https://bugzilla.kernel.org/show_bug.cgi?id=187531
http://4pda.ru/forum/index.php?showtopic=650808&view=findpost&p=55476620
http://4pda.ru/forum/index.php?showtopic=650808&view=findpost&p=55487213
you can do this，such as，modprobe silead max_x=1625 max_y=1135 invert_y=yes

ssinyagin, 2017-03-14:
@youling257 not exactly, I disable `silead_ts` module, and use gslx680_ts_acpi instead. But I will check its parameters, thanks.  Here's the link to the kernel patches: https://github.com/hopkinskong/linux_cherrytrail

ssinyagin, 2017-03-14:
@youling257 the driver does not take runtime parameters, but extracts them from the firmware header. The fwtool command line specifies these parameters as I mentioned in the ticket opening.

youling257, 2017-03-14:
do you modprobe -r silead ? or modules.blacklist silead，default load silead .
gslx680_ts_acpi、silead_ts.fw，silead.ko、mssl1680.fw

ssinyagin, 2017-03-14:
I just don't build silead module, because @hopkinskong 's patches demand that it's disabled before including `gslx680_ts_acpi`. So, silead.ko is just not present in the kernel search path.

ssinyagin, 2017-03-14:
@onitake I copied the original `SileadTouch.sys` from the MMC image, but unscramble tool produces a zero-length file. I'll send you a link privately, and will try finding one on Android partition.

ssinyagin, 2017-03-14:
good, I extracted `SileadTouch.sys` from the original Windows partition, and now the touchscreen works as it should. I will make a pull request during tomorrow, as it's quite late now :)

ssinyagin, 2017-03-14:
it needs a bit of calibration, but works fine in general

ssinyagin, 2017-03-14:
`SileadTouch.sys` contained only one firmware version. Also whatever I try, the upper ~200 pixels are not reachable by cursor.

hopkinskong, 2017-03-15:
@ssinyagin Try adjusting -w/-h param in fwtool, i just tested the two values one by one until i got a satisfied result. Then run X11 touchscreen calibration to get the best result.

onitake, 2017-03-15:
If you really, really can't get any inputs in the upper part of the screen, you can try the following:
Around line 369 in gslx680_ts_acpi.c, you can turn the dev_vdbg into a dev_warn to see each input event written to the kernel log.

If you don't see any values printed when you move around the upper part, it's likely you still have the wrong firmware.
If you do see touch events, you may be able to work around the problem by modifying axis setup around line 556. I didn't add parameters for minimum x any y values, as the allowable range is always 0..4095 and calibration can be done in userland, but in some cases, x_min and y_min may be useful.

ssinyagin, 2017-03-15:
there is input detected in the upper part pf the screen, but the cursor does not move there. I will experiment with it during the week.

samueldr, 2017-04-09:
Hi!

I extracted the silead driver from the Windows installation on my HQ64 Hi10 pro (x8350) and the firmware I got, even when using the exact same parameters as listed in the readme, works flawlessly (excluding jitter) did not even have to calibrate anything. I had an issue similar to what I believe @ssinyagin was describing with the firmware from this repo.

Before making a PR, I'd like it if @ssinyagin would try the firmware I got. 

[silead_touch_sdr-chuwi-hq64.zip](https://github.com/onitake/gslx680-acpi/files/908588/silead_touch_sdr-chuwi-hq64.zip)

In there is the windows `.sys` shipped on my device and the extracted driver.

My `fwtool` invocation used: 

```
$ perl -I gsl-firmware/tools gsl-firmware/tools/fwtool \
    -c SileadTouch.sys -3 -m 1680 -w 1260 -h 1925 -t 10 -f track,swap \
    silead_ts.fw
```

* * *

### Additional notes

> Do note that I believe that calibrating using xinput_calibrator is not needed. When the screen is rotated, the touchscreen does not know it was rotated. To rotate the touchscreen `xinput` has to be used. This is not this driver's fault, it happens on all touchscreens I tried with Linux.
> 
>     xinput --map-to-output "Silead GSLx680 Touchscreen" DSI1
> 
> The `DSI1` might be `DSI-1`, depending if the intel driver is loaded or not, xrandr will be able to tell you the name of your screens.
> 
>     xrandr | grep ^DSI | cut -f1 -d" "
> 
> This also fixes the common issues where the touchscreen is spread across connected monitors!

ssinyagin, 2017-04-09:
Samuel, unfortunately I won't be able to test this before end of April. 
So, you better issue a PR, and I will eventually test it, and rise a new issue if there's any problem.

onitake, 2022-11-22:
If you still encounter issues, please take them to https://github.com/onitake/gsl-firmware .

You're likely having a firmware issue and I'm not supporting this driver any more.

closed 2022-11-22 by onitake