onitake/gslx680-acpi #21: [feature] Windows softkey support

samueldr, 2017-04-09:
Hi,

I have a chuwi hi10 pro (HQ64/z8350), with a supported touchscreen. There is a windows button softkey that is partially used by the driver. It seems to force a click at a specific coordinate in the screen.

* * *

### Investigation

I have briefly looked at the code and it looks like there's no handling for anything else than touch. (Not an issue, as the driver is fairly generic.)

I have not xinput calibrated anything (though it shouldn't affect this print?). The firmware was extracted using this invocation:

```
$ perl -I gsl-firmware/tools gsl-firmware/tools/fwtool \
    -c SileadTouch.sys -3 -m 1680 -w 1260 -h 1925 -t 10 -f track,swap \
    silead_ts.fw
```

#### Informations gathering

Any touch on the windows softkey gives me the exact same position and pressure.
```
[ 3587.059739] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: got touch events for 1 fingers @5205
[ 3587.059772] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: touch event 0: x=672 y=1952 id=0x0 p=1
```

Keeping a finger on the screen and then touching the windows softkey does not nothing; no new event for the softkey press.

Keeping the softkey pressed and touching the screen seems to cancel the touch, but first makes a two-finger touch? On this log, I did not release the, the 2nd finger release was whilst holding the key

```
[ 3659.050157] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: got touch events for 1 fingers @15413
[ 3659.050193] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: touch event 0: x=672 y=1952 id=0x0 p=1
[ 3659.069265] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: got touch events for 1 fingers @15417
[ 3659.069298] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: touch event 0: x=672 y=1952 id=0x0 p=1
[ 3659.088048] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: got touch events for 1 fingers @15421
[ 3659.088079] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: touch event 0: x=672 y=1952 id=0x0 p=1
[ 3659.118831] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: got touch events for 1 fingers @15426
[ 3659.118866] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: touch event 0: x=672 y=1952 id=0x0 p=1
[ 3659.139449] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: got touch events for 2 fingers @15430       # <----- softkey pressed here
[ 3659.139483] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: touch event 0: x=672 y=1952 id=0x0 p=1
[ 3659.139501] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: touch event 1: x=779 y=1064 id=0x0 p=4
[ 3659.162735] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: got touch events for 2 fingers @15434       # <----- Last two-finger event
[ 3659.162772] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: touch event 0: x=672 y=1952 id=0x0 p=1
[ 3659.162790] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: touch event 1: x=756 y=1067 id=0x0 p=0
[ 3659.180567] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: got touch events for 1 fingers @15438
[ 3659.180602] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: touch event 0: x=753 y=1071 id=0x0 p=0
[ 3659.200381] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: got touch events for 1 fingers @15442
[ 3659.200416] gslx680 i2c-MSSL1680:00: gsl_ts_mt_event: touch event 0: x=753 y=1070 id=0x0 p=0
```

It looks like the hardware treats the softkey with less priority than the touchscreen. Once the touchscreen is used, anything with the softkey is ignored, until the touchscreen is not in use.

Would there be anything else useful to gather before trying to understand how to implement this?

onitake, 2017-04-10:
If the chip always generates the same coordinates for the touch key, then this should be all you need.

It's very interesting that your hardware sends a multitouch event in this case. Are you sure you need the kernel finger tracker?

Also, I probably won't accept a patch for this, since it is highly tablet dependent. It could be made configurable, but that would require much additional work.

In silead_ts (the new kernel driver), they are adding DMI based device handling AFAIK; in this case, special handling for your device could be added. If possible, try to keep the coordinate <-> key mapping in data structure so it can be implemented in a generic way.

Thanks!

samueldr, 2018-06-22:
*Closing as it works fine with jwrdegoede/linux-sunxi*

closed 2018-06-22 by samueldr