onitake/gslx680-acpi #22: Dont work multi touch on Chuwi hi 10 pro

rexido, 2017-05-07:
I have installed  this driver .... but multi touch dont work

onitake, 2017-05-07:
Have you enabled finger tracking in the firmware image?

The controller does not support finger tracking properly, so it needs to be done in the kernel driver.

rexido, 2017-05-07:
I take this firmware image https://github.com/onitake/gsl-firmware/tree/master/firmware/chuwi/hi10_pro
and this driver https://github.com/onitake/gslx680-acpi
my system is arch linux ( linux kernel 4.11 )

onitake, 2017-05-07:
Oh, I see.
This firmware should already enable finger tracking. Strange.

I don't see why it's not working, have you tried debugging the problem with evtest?

rexido, 2017-05-07:
it is one touch `nput driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x0 product 0x0 version 0x0
Input device name: "Silead GSLx680 Touchscreen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   1828
      Min        0
      Max     1925
    Event code 1 (ABS_Y)
      Value   1175
      Min        0
      Max     1260
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max     1925
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     1260
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Event: time 1494188082.994891, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 238
Event: time 1494188082.994891, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 813
Event: time 1494188082.994891, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 533
Event: time 1494188082.994891, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1494188082.994891, type 3 (EV_ABS), code 0 (ABS_X), value 813
Event: time 1494188082.994891, type 3 (EV_ABS), code 1 (ABS_Y), value 533
Event: time 1494188082.994891, -------------- SYN_REPORT ------------
Event: time 1494188083.016280, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1494188083.016280, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1494188083.016280, -------------- SYN_REPORT ------------

`

rexido, 2017-05-07:
it is multi touch 

^C[rexido@rexido-soft]: ~>$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Power Button
/dev/input/event1:      Lid Switch
/dev/input/event2:      Power Button
/dev/input/event3:      PC Speaker
/dev/input/event4:      Silead GSLx680 Touchscreen
/dev/input/event5:      Video Bus
/dev/input/event6:      LIZHICHIP USB Keyboard
/dev/input/event7:      LIZHICHIP USB Keyboard
Select the device event number [0-7]: 4
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x0 product 0x0 version 0x0
Input device name: "Silead GSLx680 Touchscreen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    813
      Min        0
      Max     1925
    Event code 1 (ABS_Y)
      Value    533
      Min        0
      Max     1260
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max     1925
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     1260
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
Event: time 1494188152.018672, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 239
Event: time 1494188152.018672, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 625
Event: time 1494188152.018672, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 816
Event: time 1494188152.018672, type 3 (EV_ABS), code 47 (ABS_MT_SLOT), value 1
Event: time 1494188152.018672, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 240
Event: time 1494188152.018672, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 749
Event: time 1494188152.018672, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 477
Event: time 1494188152.018672, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1494188152.018672, type 3 (EV_ABS), code 0 (ABS_X), value 625
Event: time 1494188152.018672, type 3 (EV_ABS), code 1 (ABS_Y), value 816
Event: time 1494188152.018672, -------------- SYN_REPORT ------------
Event: time 1494188152.042432, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 753
Event: time 1494188152.042432, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 476
Event: time 1494188152.042432, -------------- SYN_REPORT ------------
Event: time 1494188152.077163, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 477
Event: time 1494188152.077163, -------------- SYN_REPORT ------------
Event: time 1494188152.094703, type 3 (EV_ABS), code 47 (ABS_MT_SLOT), value 0
Event: time 1494188152.094703, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 624
Event: time 1494188152.094703, type 3 (EV_ABS), code 47 (ABS_MT_SLOT), value 1
Event: time 1494188152.094703, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 476
Event: time 1494188152.094703, type 3 (EV_ABS), code 0 (ABS_X), value 624
[... 77 lines cut ...]
Event: time 1494188152.668702, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 476
Event: time 1494188152.668702, type 3 (EV_ABS), code 1 (ABS_Y), value 817
Event: time 1494188152.668702, -------------- SYN_REPORT ------------
Event: time 1494188152.721862, type 3 (EV_ABS), code 47 (ABS_MT_SLOT), value 0
Event: time 1494188152.721862, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 624
Event: time 1494188152.721862, type 3 (EV_ABS), code 0 (ABS_X), value 624
Event: time 1494188152.721862, -------------- SYN_REPORT ------------
Event: time 1494188152.738724, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 625
Event: time 1494188152.738724, type 3 (EV_ABS), code 0 (ABS_X), value 625
Event: time 1494188152.738724, -------------- SYN_REPORT ------------
Event: time 1494188152.756246, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 624
Event: time 1494188152.756246, type 3 (EV_ABS), code 0 (ABS_X), value 624
Event: time 1494188152.756246, -------------- SYN_REPORT ------------
Event: time 1494188152.774243, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 625
Event: time 1494188152.774243, type 3 (EV_ABS), code 0 (ABS_X), value 625
Event: time 1494188152.774243, -------------- SYN_REPORT ------------
Event: time 1494188152.826916, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 816
Event: time 1494188152.826916, type 3 (EV_ABS), code 1 (ABS_Y), value 816
Event: time 1494188152.826916, -------------- SYN_REPORT ------------
Event: time 1494188152.853541, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1494188152.853541, type 3 (EV_ABS), code 47 (ABS_MT_SLOT), value 1
Event: time 1494188152.853541, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1494188152.853541, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1494188152.853541, -------------- SYN_REPORT ------------
^C[rexido@rexido-soft]: ~>$

onitake, 2017-05-08:
That looks correct to me, at least from the perspective of the input subsystem.

Your problem must be elsewhere. Maybe your Xorg is misconfigured, or your applications don't support multitouch?
I'm not an expert on how multitouch works on the userspace side. You need to ask somewhere else. Sorry.

pepelillo, 2017-05-09:
Hi.
RExido, so silead-acpi driver works on 4.11 but multitouch?
Thanks

UwUnyaa, 2022-07-03:
I'm having the same issue with a different device (that I'm about to make a pull request with firmware for). Enabling software finger tracking doesn't seem to fix the issue. I get only one touch point at a time no matter what.