onitake/gslx680-acpi #23: Touchscreen not working after installing the driver

fl0uf, 2017-06-12:
Hello,

i've extracted the firmware of my MPMAN CONVERTER 8 and after compiling your driver, the module is inserted correctly (see the dmesg.txt). I've launched evtest and see the touchscreen device and when I choose it, nothing happens even after restarting X.

My operating system is Ubuntu Artful 17.10 with a 4.12 rc4 kernel (to support the wifi/BT driver).
[dmesg.txt](https://github.com/onitake/gslx680-acpi/files/1069662/dmesg.txt)

Have you got an idea ?

onitake, 2017-06-12:
Can you run `evtest` on the HID device? Does that generate touch events?

fl0uf, 2017-06-21:
The device is not recognized with the HID driver, i have an error -1 when the silead driver loads, i think it is in the dmesg.txt, only your driver detects the device under X.

onitake, 2017-06-21:
Your dmesg contains messages for three successful driver loads and a lot of garbage.

If loading fails with an error, please post only the relevent messages, not the whole file.