onitake/gslx680-acpi #27: New firmware needed... Teclast X98 II Plus...

satchboogie67, 2018-03-23:
Greatings,
 
Please, would it be possible to extract a firmware from my Windows 10 ".sys" driver ?
I don't really feel I can do that by myself, so I would really appreciate a little help, if possible... 
I really wish to use the tablet with ubuntu. Everything works great except the touchscreen... The screen by itself is 1536 x 2048 pixels.

I hope you will be able to take a look at this request. I thank you so much in advance.

[SileadTouch.zip](https://github.com/onitake/gslx680-acpi/files/1842912/SileadTouch.zip)

onitake, 2018-03-28:
That file seems a bit small: 213 bytes

Can you try to upload it again?

onitake, 2018-03-30:
Here's a howto for extracting firmware: https://github.com/onitake/gsl-firmware/blob/master/firmware/trekstor/surftab-twin-10.1-ST10432-8/README.md#command-to-find-the-offsets-used-for-extraction

And here's the result: [teclast-x98-ii-plus.zip](https://github.com/onitake/gslx680-acpi/files/1863207/teclast-x98-ii-plus.zip)

Once you've tested it, can you submit the firmware on https://github.com/onitake/gsl-firmware ?
Thanks!

closed 2022-11-22 by onitake