onitake/gslx680-acpi #32: Much more confuse

springhack, 2019-03-26:
I'm pretty sure my touchscreen is gslx680, because I got `gslX680` on `i2c-5` in my Android OS;
But in lubuntu I got this:
```
dosk@Bullshit:~$ sudo find /sys/ | grep 1680
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/uevent
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power/runtime_active_time
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power/runtime_active_kids
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power/runtime_usage
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power/runtime_status
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power/autosuspend_delay_ms
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power/async
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power/runtime_suspended_time
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power/runtime_enabled
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power/control
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power_resources_D0
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power_resources_D0/LNXPOWER:16
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/power_resources_D0/LNXPOWER:17
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/adr
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/hid
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/path
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/subsystem
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/status
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/GSL1680:00/modalias
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/uevent
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power/runtime_active_time
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power/runtime_active_kids
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power/runtime_usage
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power/runtime_status
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power/autosuspend_delay_ms
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power/async
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power/runtime_suspended_time
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power/runtime_enabled
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power/control
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power_resources_D0
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power_resources_D0/LNXPOWER:16
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/power_resources_D0/LNXPOWER:17
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/adr
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/hid
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/path
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/subsystem
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/status
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/808622C1:05/MSSL1680:00/modalias
/sys/bus/acpi/devices/GSL1680:00
/sys/bus/acpi/devices/MSSL1680:00
```
and 
```
dosk@Bullshit:~$ ls -al /sys/bus/i2c/devices/
total 0
drwxr-xr-x 2 root root 0 3月  27 03:11 .
drwxr-xr-x 4 root root 0 3月  27 02:05 ..
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-0 -> ../../../devices/pci0000:00/808622C1:00/i2c-0
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-1 -> ../../../devices/pci0000:00/808622C1:01/i2c-1
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-10 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-10
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-11 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-11
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-12 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-12
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-13 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/i2c-13
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-14 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-2/i2c-14
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-2 -> ../../../devices/pci0000:00/808622C1:02/i2c-2
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-3 -> ../../../devices/pci0000:00/808622C1:03/i2c-3
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-4 -> ../../../devices/pci0000:00/808622C1:04/i2c-4
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-5 -> ../../../devices/pci0000:00/808622C1:05/i2c-5
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-6 -> ../../../devices/pci0000:00/808622C1:06/i2c-6
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-7 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-7
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-8 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-8
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-9 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-9
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-GDIX1001:00 -> ../../../devices/pci0000:00/808622C1:05/i2c-5/i2c-GDIX1001:00
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-HIMX2051:00 -> ../../../devices/pci0000:00/808622C1:00/i2c-0/i2c-HIMX2051:00
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-INT33F4:00 -> ../../../devices/pci0000:00/808622C1:06/i2c-6/i2c-INT33F4:00
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-KIOX0009:00 -> ../../../devices/pci0000:00/808622C1:02/i2c-2/i2c-KIOX0009:00
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-OVTI2680:00 -> ../../../devices/pci0000:00/808622C1:00/i2c-0/i2c-OVTI2680:00
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-TXN24292:00 -> ../../../devices/pci0000:00/808622C1:00/i2c-0/i2c-TXN24292:00
lrwxrwxrwx 1 root root 0 3月  27 02:05 i2c-TXN27501:00 -> ../../../devices/pci0000:00/808622C1:00/i2c-0/i2c-TXN27501:00
```
so the driver won't match my device, and in my android I can't search any file contain `680` or `silead` in file name, so much more confuse ...

onitake, 2019-03-28:
MSSL1680 is the ACPI ID of all Silead touchscreen controllers.
That you have also have GSL1680 and a GDIX1001 in your device list is really confusing.

Can you get a dump of your DSDT and send it to me?
See here for instructions: https://wiki.archlinux.org/index.php/DSDT

springhack, 2019-03-29:
@onitake Sure, here it is, thanks a lot
```
dosk@Bullshit:~$ iasl -d dsdt.dat

Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20180629
Copyright (c) 2000 - 2018 Intel Corporation

Input file dsdt.dat, Length 0x1F714 (128788) bytes
ACPI: DSDT 0x0000000000000000 01F714 (v02 ALASKA A M I    01072009 INTL 20120913)
Pass 1 parse of [DSDT]
Pass 2 parse of [DSDT]
Parsing Deferred Opcodes (Methods/Buffers/Packages/Regions)

Parsing completed

Found 2 external control methods, reparsing with new information
Pass 1 parse of [DSDT]
Pass 2 parse of [DSDT]
Parsing Deferred Opcodes (Methods/Buffers/Packages/Regions)

Parsing completed
Disassembly completed
ASL Output:    dsdt.dsl - 1035947 bytes

 iASL Warning: There were 2 external control methods found during
 disassembly, but only 0 were resolved (2 unresolved). Additional
 ACPI tables may be required to properly disassemble the code. This
 resulting disassembler output file may not compile because the
 disassembler did not know how many arguments to assign to the
 unresolved methods. Note: SSDTs can be dynamically loaded at
 runtime and may or may not be available via the host OS.

 To specify the tables needed to resolve external control method
 references, the -e option can be used to specify the filenames.
 Example iASL invocations:
     iasl -e ssdt1.aml ssdt2.aml ssdt3.aml -d dsdt.aml
     iasl -e dsdt.aml ssdt2.aml -d ssdt1.aml
     iasl -e ssdt*.aml -d dsdt.aml

 In addition, the -fe option can be used to specify a file containing
 control method external declarations with the associated method
 argument counts. Each line of the file must be of the form:
     External (<method pathname>, MethodObj, <argument count>)
 Invocation:
     iasl -fe refs.txt -d dsdt.aml
```
[dsdt.dsl.zip](https://github.com/onitake/gslx680-acpi/files/3020793/dsdt.dsl.zip)

onitake, 2019-03-29:
Ok, so here's some observations:
- There's a MSSL1680 device, defined as usual. The _STA method (which reports if a device is enabled or not) returns True (0F) based on two condition variables: BDID and OSID, which I assume are Board ID and Operating System ID. They have different values than the expected ones on your system, so this device is disabled.
- There's also an extra GSL1680 device, but it's always disabled (the _STA method returns 0)
- Then, there's a GDIX1001 device, which is always enabled (_STA returns 0F). This represents a Goodix controller.
- All three devices are on bus I2C6, and use the same GPIO lines for power control, but have different addresses: The Silead has 0x40, while the Goodix has 0x14.

From that, I conclude that you must have a Goodix controller in your device, which is not supported by this driver.