onitake/gslx680-acpi #33: make not working (Ubuntu 21.04)

incuubus, 2021-06-07:
Hello,

I am trying to run make in the gslx680-acpi folder but I get this error:

Skipping BTF generation for /tmp/gslx680-acpi/gslx680_ts_acpi.ko due to unavailability of vmlinux

Does anyone know how i can solve this error?
I am using this compiler: gcc (Ubuntu 10.3.0-1ubuntu1) 10.3.0

onitake, 2021-06-08:
Is this really an error?
From my understanding, it's only a warning that is printed when trying to build out-of-tree kernel modules without building the whole kernel first.
The resulting module should still work, the BTF isn't really needed here.

For reference:
* https://www.spinics.net/lists/bpf/msg30651.html
* https://lkml.org/lkml/2020/11/16/1452
* https://www.kernel.org/doc/html/latest/bpf/btf.html

incuubus, 2021-06-08:
Thanks you alot for the quick answer =)
I'm sorry to ask these silly questions. I'm quite new to linux and wanted to install it on my chuwi hi10 because android and windows don't work on it anymore and can't be reinstalled (or at least it doesn't seem to work for me). But it is just too confusing for me =/

I guess maybe the problem is, that the firmware of hi10pro is not compatible on just the hi10.
I have to copy the firmware "silead_ts.fw" to /lib/firmware/silead right? How do I have to name it? I've read somewhere that the name must be "mssl0017.fw".

I'm trying to install it with the steps described here: https://github.com/willyneutron/lubuntu_in_chuwi_Hi10Pro

onitake, 2021-06-08:
The firmware name is hardcoded to silead_ts.fw, as you can see here: https://github.com/onitake/gslx680-acpi/blob/master/gslx680_ts_acpi.c#L66

mssl0017.fw might work with some other drivers, but not gslx680-acpi.

Although, please note that I'm not supporting gslx680-acpi any more. The correct way is to use the silead driver that is part of the Linux kernel. Unfortunately, this driver requires a bit more work to add support for new hardware, but when a device is supported, you can simply install the firmware from here and it will work automatically: https://github.com/onitake/gsl-firmware/tree/master/firmware/linux

closed 2023-01-04 by onitake