onitake/gslx680-acpi #36: make not working

fly206, 2023-01-03:
Hello,

I'm trying to run make and get an error:
```
gslx680-acpi-master/gslx680_ts_acpi.c:753:19: error: initialization of ‘void (*)(struct i2c_client *)’ from incompatible pointer type ‘int (*)(struct i2c_client *)’ [-Werror=incompatible-pointer-types]
  753 |         .remove = gsl_ts_remove,
```
My kernel is:
```
Linux pctest 6.1.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 21 Dec 2022 22:27:55 +0000 x86_64 GNU/Linux
```
Then I modified the code:
```diff
diff -urN a/gslx680_ts_acpi.c b/gslx680_ts_acpi.c
--- a/gslx680_ts_acpi.c	2016-04-28 21:12:01.000000000 +0800
+++ b/gslx680_ts_acpi.c	2023-01-03 11:23:25.167129379 +0800
@@ -663,10 +663,9 @@
 	return 0;
 }

-int gsl_ts_remove(struct i2c_client *client) {
+void gsl_ts_remove(struct i2c_client *client) {
 	/* Power the device off */
 	gsl_ts_power(client, true);
-	return 0;
 }

 static int __maybe_unused gsl_ts_suspend(struct device *dev)
```
After that, it worked fine.

onitake, 2023-01-03:
Thanks, but this is a duplicate of PR #34 .

If you can, please send a patch that works with both kernel 6.1+ and older kernels.

onitake, 2023-01-04:
Fixed in #37 , thank you!

closed 2023-01-04 by onitake